wrenger/bitfield-struct-rs #1: Defining a 1-bit non bool member fails

damageboy, 2022-04-21:
The following sample:

```rust
use bitfield_struct::bitfield;

#[bitfield(u64)]
struct PageTableEntry {
    /// defaults to 32 bits for u32
    addr: u32,

    /// public field -> public accessor functions
    #[bits(12)]
    pub size: usize,

    /// padding: No accessor functions are generated for fields beginning with `_`.
    #[bits(5)]
    _p: u8,

    #[bits(1)]
    bug: u8,

    /// interpreted as 1 bit flag
    present: bool,

    /// sign extend for signed integers
    #[bits(13)]
    negative: i16,
}
```

Will fail with:

```shell
  --> src/main.rs:3:1
   |
3  | #[bitfield(u64)]
   | ^^^^^^^^^^^^^^^^ expected `u8`, found `bool`
...
17 |     bug: u8,
   |          -- expected `u8` because of return type
   |
   = note: this error originates in the attribute macro `bitfield` (in Nightly builds, run with -Z macro-backtrace for more info)

For more information about this error, try `rustc --explain E0308`.
```

wrenger, 2022-05-05:
Oh, I will look into that. 
I previously only tested members with 1 bit size for the type bool.

wrenger, 2022-05-05:
Fixed by: https://github.com/wrenger/bitfield-struct-rs/commit/71936eec0594a82ece52c7d8b56bf214740e0c22

closed 2022-05-05 by wrenger