wrenger/bitfield-struct-rs #2: Markers are unsupported

adenine-dev, 2022-07-14:
The following sample
```rust
use std::marker::PhantomData;

#[bitfield(u32)]
pub struct Foo<T> {
    _unused: u32,

    _marker: PhantomData<T>,
}
```
Fails to compile with the following error:
```rust
error: unsupported type
  --> main.rs:9:14
  |
9 |     _marker: PhantomData<T>,
  |              ^^^^^^^^^^^^^^
```

While some types should indeed be unsupported, markers probably should be supported, maybe even all zero size types? Not sure how one would go about fixing this other than potentially just extending `type_bits` to support marker types, but this disqualifies user defined markers, so doesn't seem like the best solution.

wrenger, 2022-07-20:
Well yes, maybe the best idea would be to just support zero sized types for fields without accessors (prefixed with `_`).
However, it is somewhat difficult to find out the size of a type in a procedural macro, as it has no access to detailed type information (only basic Tokens). With generics this becomes even more difficult...

wrenger, 2022-12-22:
Zero-sized members are now supported (prefixed by `#[bits(0)]`). They are ignored by the bitfield generation.
Generics, however, are still difficult. I might look into that in the future, but I can't promise.