wrenger/bitfield-struct-rs #4: 0.1.8's Debug trait impl is a breaking change

jstarks, 2022-12-18:
I'm very happy that Debug is now generated, but this is a breaking change--I have types where I have manually implemented Debug that now fail to compile. The semver version should have changed to 0.2.0.

Separately, it would be nice to have an opt-out on a per-type basis, maybe via an attribute. Sometimes I want a custom Debug impl.

wrenger, 2022-12-22:
Yes, that are valid points.

Making Debug opt-out is a good idea. For that, we need a good way of specifying that. 
Somewhat along the following maybe:
```rs
#[bitfield(u64, NoDebug)]
struct PT {
   //...
}
```

wrenger, 2022-12-23:
Ok I've implemented this as part of 0.2.0: 2e5faf81c00f0d6173d134fddef2bce24dc845ba

closed 2023-01-04 by jstarks