wrenger/bitfield-struct-rs #9: how to put a small bitfield_struct into a big bitfield_struct

zilongmix, 2023-03-06:
```rust
#[bitfield[u8]]
pub struct PMPcfgIn {
    pub r: bool,
    pub w: bool,
    pub x: bool,
    #[bits(2)]
    pub a: u8,
    #[bits(2)]
    _pad: u8,
    pub l: bool,
}

#[bitfield[u64]]
pub struct PMPcfg {
    pub pmp0cfg: PMPcfgIn, // err, unsupported type
    pub pmp1cfg: u8,
    pub pmp2cfg: u8,
    pub pmp3cfg: u8,
    pub pmp4cfg: u8,
    pub pmp5cfg: u8,
    pub pmp6cfg: u8,
    pub pmp7cfg: u8,
}
```
I just want to replace `u8` with `PMPcfgIn` in the `PMPcfg` structure, is there any way to do that?

wrenger, 2023-03-06:
At the moment, PMPcfgIn has to implement From/Into u64 (the type behind PMPcfg).
You can do this manually. 

Also you have to explicitly specify the bit size (using #[bits(8)] for example)

zilongmix, 2023-03-07:
```rust
#[bitfield(u8)]
pub struct PMPcfgIn {
    pub r: bool,
    pub w: bool,
    pub x: bool,
    #[bits(2)]
    pub a: u8,
    #[bits(2)]
    _pad: u8,
    pub l: bool,
}

impl From<u64> for PMPcfgIn {
    fn from(value: u64) -> Self {
        PMPcfgIn::from(value as u8)
    }
}

impl From<PMPcfgIn> for u64 {
    fn from(val: PMPcfgIn) -> Self {
        val.0 as u64
    }
}

#[bitfield(u64)]
pub struct PMPcfg {
    #[bits(8)]
    pub pmp0cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp1cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp2cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp3cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp4cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp5cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp6cfg: PMPcfgIn,
    #[bits(8)]
    pub pmp7cfg: PMPcfgIn,
}
```
Thanks, that works for me!

closed 2023-03-10 by wrenger

pwfff, 2023-07-09:
seems like this is no longer the case after `from_bits` and `into_bits` were added? having trouble implementing those. would be nice if they automatically came with `#[bitfield(...)]`.

```
#[bitfield(u8)]
pub struct SPI_STATUS {
    pub stst: bool,
    pub olb: bool,
    pub ola: bool,
    pub s2gb: bool,
    pub s2ga: bool,
    pub otpw: bool,
    pub ot: bool,
    pub stallguard: bool,
}

impl SPI_STATUS {
    // This has to be a const fn
    const fn into_bits(self) -> u32 {
        self.int() as u32
    }
    const fn from_bits(value: u32) -> Self {
        Self::new().with_int(value as u8)
    }
}
```
```
error[E0599]: no method named `int` found for struct `SPI_STATUS` in the current scope
   --> src/registers/tmc2240/mod.rs:763:14
    |
748 | #[bitfield(u8)]
    | --------------- method `int` not found for this struct
...
763 |         self.int() as u32
    |              ^^^ method not found in `SPI_STATUS`

error[E0599]: no method named `with_int` found for struct `SPI_STATUS` in the current scope
   --> src/registers/tmc2240/mod.rs:766:21
    |
748 | #[bitfield(u8)]
    | --------------- method `with_int` not found for this struct
...
766 |         Self::new().with_int(value as u8)
    |                     ^^^^^^^^ help: there is a method with a similar name: `with_ot`
```