wrenger/bitfield-struct-rs #12: A way to specify bitfield ordering

OliverGavin, 2023-04-26:
The docs state that "The first field occupies the least significant bits".

I have the following snippet from a packet I am trying to parse;
```
// 0101 .... = length: 5
// .... 1... = a: true
// .... .0.. = b: false
// .... ..01 = flags: 1
```

Intuitively I would write the bitfield struct in the following order;
```
    #[bits(4)]
    pub len: u8,
    #[bits(1)]
    pub a: bool,
    #[bits(1)]
    pub b: bool,
    #[bits(2)]
    pub flags: u8,
```

But that is incorrect as the first field in my struct occupies the most significant bits.

Bellow is an example from the bitfield crate which seems to use the MSB first;

```
bitfield!{
    struct IpV4Header(MSB0 [u8]);
    u32;
    get_version, _: 3, 0;
    get_ihl, _: 7, 4;
    get_dscp, _: 13, 8;
    get_ecn, _: 15, 14;
    get_total_length, _: 31, 16;
    get_identification, _: 47, 31;
    get_df, _: 49;
    get_mf, _: 50;
    get_fragment_offset, _: 63, 51;
    get_time_to_live, _: 71, 64;
    get_protocol, _: 79, 72;
    get_header_checksum, _: 95, 79;
    get_source_address, _: 127, 96;
    get_destination_address, _: 159, 128;
}
```

Would it be possible/practical to implement in this crate?

wrenger, 2023-04-26:
Starting with the least significant bits is a design decision of this crate. You would have to reverse the order of the members in this case.

I currently see little value in making the field order configurable. While it might help some, it also makes this crate more complex and confusing...

OliverGavin, 2023-04-26:
I understand - it makes sense. I just preferred your API over the `bitfield` example above (which does support MSB ordering). But I can probably live with the minor ugliness :p

Your crate seems to do a very good job at what it is supposed to do :)

wrenger, 2023-04-26:
> Your crate seems to do a very good job at what it is supposed to do :)

Thank :)

> I'm not sure how feasible it is but would you be open to a PR that takes MSB0 or LSB0 in the macro and uses https://docs.rs/bitorder/latest/bitorder/ internally?

If you want, you could open a PR yourself. Just fork the repo, and afterward, a button for a new PR should be visible in your fork.

pwfff, 2023-07-10:
oops... implemented this for myself before i noticed this issue:

```diff
diff --git a/src/lib.rs b/src/lib.rs
index 690ba43..d608a81 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -251,10 +251,11 @@ fn bitfield_inner(args: TokenStream, input: TokenStream) -> syn::Result<TokenStr
         return Err(syn::Error::new(span, "only named fields are supported"));
     };
 
+    let be = true;
     let mut offset = 0;
     let mut members = Vec::with_capacity(fields.named.len());
     for field in fields.named {
-        let f = Member::new(ty.clone(), field, offset)?;
+        let f = Member::new(ty.clone(), field, offset, be)?;
         offset += f.bits;
         members.push(f);
     }
@@ -343,6 +344,7 @@ fn bitfield_inner(args: TokenStream, input: TokenStream) -> syn::Result<TokenStr
 
 /// Represents a member where accessor functions should be generated for.
 struct Member {
+    be: bool,
     offset: usize,
     bits: usize,
     base_ty: syn::Type,
@@ -360,7 +362,7 @@ struct MemberInner {
 }
 
 impl Member {
-    fn new(base_ty: syn::Type, f: syn::Field, offset: usize) -> syn::Result<Self> {
+    fn new(base_ty: syn::Type, f: syn::Field, offset: usize, be: bool) -> syn::Result<Self> {
         let span = f.span();
 
         let syn::Field {
@@ -398,6 +400,7 @@ impl Member {
             attrs.retain(|a| !a.path().is_ident("bits"));
 
             Ok(Self {
+                be,
                 offset,
                 bits,
                 base_ty,
@@ -417,6 +420,7 @@ impl Member {
             }
 
             Ok(Self {
+                be,
                 offset,
                 bits,
                 base_ty,
@@ -453,6 +457,7 @@ impl Member {
 impl ToTokens for Member {
     fn to_tokens(&self, tokens: &mut TokenStream) {
         let Self {
+            be,
             offset,
             bits,
             base_ty,
@@ -462,6 +467,11 @@ impl ToTokens for Member {
             return Default::default();
         };
 
+        let offset_def = match *be {
+            true => quote! { (#base_ty::BITS as usize) - #offset - #bits },
+            false => quote! { #offset },
+        };
+
         let ident_str = ident.to_string();
 
         let with_ident = format_ident!("with_{ident}");
@@ -482,7 +492,7 @@ impl ToTokens for Member {
 
         let code = quote! {
             const #bits_ident: usize = #bits;
-            const #offset_ident: usize = #offset;
+            const #offset_ident: usize = #offset_def;
 
             #doc
             #[doc = #location]
@@ -493,12 +503,12 @@ impl ToTokens for Member {
                 };
                 #[allow(unused_comparisons)]
                 debug_assert!(value <= #mask, "value out of bounds");
-                Self(self.0 & !(#mask << #offset) | (value & #mask) << #offset)
+                Self(self.0 & !(#mask << Self::#offset_ident) | (value & #mask) << Self::#offset_ident)
             }
             #doc
             #[doc = #location]
             #vis const fn #ident(&self) -> #ty {
-                let this = (self.0 >> #offset) & #mask;
+                let this = (self.0 >> Self::#offset_ident) & #mask;
                 #from
             }
             #doc
```

i was planning on making a PR with endianness configurable with `be = true`.
is there an advantage to using `bitorder`?

OliverGavin, 2023-07-11:
@pwfff please do, these macros are a bit cleaner and easier to use 😀

pwfff, 2023-07-11:
gave it a shot: https://github.com/wrenger/bitfield-struct-rs/pull/19

closed 2023-08-01 by wrenger