wrenger/bitfield-struct-rs #14: Positive signed integers fail

tomasbarton-stemcell, 2023-05-16:
The following test fails when added to `/tests/test.rs`:

```rust
#[test]
fn positive_signed() {
    #[bitfield(u32)]
    struct MyBitfield {
        negative: i32,
    }

    let v1 = MyBitfield::new().with_negative(-3);
    assert_eq!(v1.negative(), -3);

    let v2 = MyBitfield::new().with_negative(0);
    assert_eq!(v2.negative(), 0);

    let v3 = MyBitfield::new().with_negative(3);
    assert_eq!(v3.negative(), 3);
}
```

with:

```
---- positive_signed stdout ----
thread 'positive_signed' panicked at 'assertion failed: value <= 0xffffffff', tests/test.rs:125:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

The problem is that the condition of the `debug_assert` here 

https://github.com/wrenger/bitfield-struct-rs/blob/ff0107b1c237348ca266d9a1fe2d041dde093f75/src/lib.rs#L469

does not work for signed integers. The mask will be a hexadecimal number which rust interprets "literally" (ie the condition is `value< = 0xfffffff` which Rust sees as `value <= -1`)

wrenger, 2023-05-17:
Yes, the mask overflows. The check generally does not cover signed integers very well, as it does not detect if an integer is too small.

wrenger, 2023-05-17:
I've improved the assert in 0.4.2 (https://github.com/wrenger/bitfield-struct-rs/commit/ceefaaa49b60b45ed7ab33ed3fb513b48fd17ca3)

closed 2023-05-19 by wrenger