wrenger/bitfield-struct-rs #22: Struct that fully contains a signed integer causes clippy error.

snakehand, 2023-10-25:
This struct produces a clippy error:

`
#[bitfield(u16)]
/// X axis output value
pub struct OutX{
    #[bits(16)]
    val: i16,
}
`

error: &-masking with zero
   --> src/lib.rs:564:1
    |
564 | #[bitfield(u16)]
    | ^^^^^^^^^^^^^^^^
    |
    = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#bad_bit_mask
    = note: `#[deny(clippy::bad_bit_mask)]` on by default
    = note: this error originates in the attribute macro `bitfield` (in Nightly builds, run with -Z macro-backtrace for more info)

I have debugged it and tracked it down to a debug assert, and propse this change that captures the same logic without causing errors in clippy:

`
index d0e0c39..669947b 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -727,7 +727,7 @@ fn parse_field(attrs: &[syn::Attribute], ty: &syn::Type, ignore: bool) -> syn::R
             // Bounds check and remove leading ones from negative values
             ret.into = quote! {{
                 #[allow(unused_comparisons)]
-                debug_assert!(if this >= 0 { this & !#mask == 0 } else { !this & !#mask == 0 }, "value out of bounds");
+                debug_assert!(if this >= 0 { this | #mask == #mask } else { !(this | #mask) == 0 }, "value out of bounds");
                 (this & #mask) as _
             }};
         }
`

snakehand, 2023-10-29:
[bitfield_err.zip](https://github.com/wrenger/bitfield-struct-rs/files/13199447/bitfield_err.zip)

Minimal example

wrenger, 2023-11-03:
Hi @snakehand, thanks for your PR. I'm currently traveling but I'll look into this as soon as I'm back next week.

Can you also tell me the rustc/Clippy version you are using?

wrenger, 2023-11-11:
The bounds check was error-prone and has been fixed (as discussed in #24). This also resolves this issue.

commit 8de20781 (2023-11-11): :bug: Fix bounds check for signed types #24

Coincidentally, this also fixes #22
closed 2023-11-11 by wrenger