wrenger/bitfield-struct-rs #24: panic with 32 bit signed int

jerrinot, 2023-11-10:
Thanks for this cool macro! 

This code is panicking with `value out of bounds`:
```rust
#[test]
fn negative_signed() {
    #[bitfield(u64)]
    struct MyBitfield {
        negative: i32,
        #[bits(32)]
        __: u32,
    }

    let _ = MyBitfield::new().with_negative(-1);
}
```

When I change it to
```
#[test]
fn negative_signed() {
    #[bitfield(u32)]
    struct MyBitfield {
        negative: i32,
    }

    let f = MyBitfield::new().with_negative(-1);
    assert_eq!(-1, f.negative())
}
```
then it's working fine.

jerrinot, 2023-11-10:
I've been looking at this for a bit. It seems the root cause is in signed casting which extends the signed bit. 
 
This demonstrates the idea: https://github.com/wrenger/bitfield-struct-rs/compare/main...jerrinot:bitfield-struct-rs:sign_shifting?expand=1#diff-b1a35a68f14e696205874893c07fd24fdb88882b47c23cc0e0c80a30c7d53759R733-R761

I am fairly sure there is a better fix, I have no idea what am I doing :)

wrenger, 2023-11-11:
Oh yes, thanks for catching this. This happens when the fields have the same bit size as their signed type. I've redesigned the bounds check accordingly in [8de2078](https://github.com/wrenger/bitfield-struct-rs/commit/8de20781e87fd8dc0d71e6a1b10c08c953dd8fb1), which will land in 5.5.

closed 2023-11-11 by wrenger

jerrinot, 2023-11-12:
many thanks for a super quick fix and release! I just included the latest version in my project and it works like a charm!