wrenger/bitfield-struct-rs #26: Converters instead of traits to support external types

Chaoses-Ib, 2023-11-11:
Right now, custom types can only be supported by implementing `into_bits` and `from_bits` or `From` and `Into`. Since we can't implement methods for an external type, it has to be wrapped as another type to be used in a bitfield struct. For example:

```rust
enum CustomEnum {
    A = 0,
    B = 1,
    C = 2,
}

struct OptionalCustomEnum(Option<CustomEnum>);

impl OptionalCustomEnum {
    const fn into_bits(self) -> u32 {
        todo!()
    }

    const fn from_bits(value: u32) -> Self {
        todo!()
    }
}

#[bitfield(u32)]
struct MyBitfield {
    #[bits(16)]
    custom: CustomEnum,

    #[bits(16)]
    __: u32,
}
```
This makes the operations inconvenient:
```rust
let mut val = MyBitfield::new().with_custom(OptionalCustomEnum(CustomEnum::A));
val.set_custom(OptionalCustomEnum(CustomEnum::A));
let custom = val.custom().0;

// or impl From<Option<CustomEnum>> for OptionalCustomEnum and:
let mut val = MyBitfield::new().with_custom(CustomEnum::A.into());
val.set_custom(CustomEnum::A.into());
let custom = val.custom().0;
```

If the conversion can be specified through converters instead of traits, for example:
```rust
enum CustomEnum {
    A = 0,
    B = 1,
    C = 2,
}

struct CustomEnumConverter;

impl CustomEnumConverter {
    const fn into_bits(value: CustomEnum) -> u32 {
        todo!()
    }

    const fn from_bits(value: u32) -> CustomEnum {
        todo!()
    }
}

#[bitfield(u32)]
struct MyBitfield {
    #[bits(16, converter = CustomEnumConverter)]
    custom: CustomEnum,

    #[bits(16)]
    __: u32,
}
```

The operations can be simpler and more intuitive:

```rust
let mut val = MyBitfield::new().with_custom(CustomEnum::A);
val.set_custom(CustomEnum::A);
let custom = val.custom();
```

wrenger, 2023-11-13:
Good point. You can provide your own converter functions via:

```rust
#[derive(Debug, PartialEq, Eq)]
#[repr(u16)]
enum CustomEnum {
    A = 0,
    B = 1,
    C = 2,
}

const fn my_from_bits(value: u16) -> CustomEnum {
    match value {
        0 => CustomEnum::A,
        1 => CustomEnum::B,
        _ => CustomEnum::C,
    }
}

const fn my_into_bits(value: CustomEnum) -> u16 {
    value as _
}

#[bitfield(u16)]
#[derive(PartialEq, Eq)]
struct MyBitfield {
    /// Supports any type, with default/to/from expressions (that are const eval)
    /// - into/from call #ty::into_bits/#ty::from_bits if nothing else is specified
    #[bits(14, default = CustomEnum::B, from = my_from_bits, into = my_into_bits)]
    custom: CustomEnum,
    #[bits(2)]
    __: (),
}

let val = MyBitfield::new();
assert_eq!(val.custom(), CustomEnum::B);
```

Chaoses-Ib, 2023-11-14:
Oh, I've seen this in readme, but I scanned too fast to think it means `From<u32>::from` and `Into<CustomEnum>::into`... Thanks for your help.

closed 2023-11-14 by Chaoses-Ib