wrenger/bitfield-struct-rs #36: Overflow tests failing in Release mode

osteffenrh, 2024-04-12:
The tests `negative_pos_overflow` and `negative_neg_overflow` are failing in release mode, while passing otherwise.

```
$ cargo test -r
[...]
---- negative_pos_overflow stdout ----
note: test did not panic as expected
---- negative_neg_overflow stdout ----
note: test did not panic as expected

failures:
    negative_neg_overflow
    negative_pos_overflow

test result: FAILED. 13 passed; 2 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.00s

error: test failed, to rerun pass `--test test`
```

```
$ cargo test
[...]
test result: ok. 7 passed; 0 failed; 0 ignored; 0 measured; 0 filtered out; finished in 0.18s
```

Version 0.6, commit 1f8620e

Toolchains tried:
- nightly-x86_64-unknown-linux-gnu, rustc 1.79.0-nightly (a07f3eb43 2024-04-11)
- stable-x86_64-unknown-linux-gnu, rustc 1.77.2 (25ef9e3d8 2024-04-09)

wrenger, 2024-04-14:
Interesting, I didn't catch that. Could be because debug_assert is disabled.

wrenger, 2024-04-14:
8311480213d970b45c3900a03f61e1278f2b0529: Disables these tests if we have no debug_asserts. The expected behavior is that all bits exceeding the field size are cut off.

closed 2024-04-14 by wrenger