wrenger/bitfield-struct-rs #38: Support custom inner types

mkroening, 2024-06-01:
Hi Lars, thanks for the great library!

I'd like to define types from virtio with defined endianness such as:

```c
le32 { 
  vqn : 16; 
  next_off : 15; 
  next_wrap : 1; 
};
```

See [Virtual I/O Device (VIRTIO) Version 1.2 — 1.4 Structure Specifications](https://docs.oasis-open.org/virtio/virtio/v1.2/cs01/virtio-v1.2-cs01.html#x1-80004) for details on the notation.

It would be great if this crate allowed to change the storage type of the struct (independently of the number type used for internal bitshift operations).

I'd envision the macro to work similar to this:

```rust
#[bitfield(u32, inner = le32, from = le32::from_ne, into = le32::to_ne)]
pub struct NotificationData {
    pub vqn: u16,
    #[bits(15)]
    pub next_off: u16,
    pub next_wrap: bool,
}
```

`le32` is from the [`endian-num`](https://docs.rs/endian-num) crate.

wrenger, 2024-06-02:
@mkroening, I think from a technical standpoint, this is fine. However, I'm a bit unsure if this should be really part of the bitfield itself or if a wrapper type might be a better idea.

On the other hand, since `Deref`/`DerefMut` doesn't work here, the wrapper might be quite verbose as shown below.

```rust
#[bitfield(u32)]
#[derive(PartialEq, Eq)]
struct MyBitfield {
    data: u32,
}

let mut my_bitfield = CustomInner::<MyBitfield>::new(MyBitfield::new());

my_bitfield.set(MyBitfield::new().with_data(1));
assert_eq!(my_bitfield.get().data(), 1);

my_bitfield.with(|b| b.with_data(2));
assert_eq!(my_bitfield.get().data(), 2);

// wrapper type

#[derive(PartialEq, Eq, Clone, Copy, Debug)]
#[repr(transparent)]
struct CustomInner<T>(u32, PhantomData<T>);

impl<T: From<u32> + Into<u32>> CustomInner<T> {
    fn new(val: T) -> Self {
        Self(val.into(), PhantomData)
    }
    fn get(&self) -> T {
        self.0.into()
    }
    fn set(&mut self, val: T) {
        self.0 = val.into()
    }
    fn with(&mut self, f: impl FnOnce(T) -> T) {
        self.set(f(self.get()))
    }
}
```

Still, a plus of your PR is that it makes it possible to specify endianness.

mkroening, 2024-06-02:
Thanks for the swift reply!

This kind of wrapper would kind of defeat the purpose of my proposal, I think. Today, I can already manually convert the `le32` field to and from `u32` and then again to and from `MyBitfield` for operations. Your proposed wrapper would remove one conversion, but would still require explicit conversion from the wrapper to the bitfield while adding an additional type and obscuring the underlying `le32`, that I'd like to expose.

My proposal would allow directly operating on `MyBitfield` with the correct byteorder. This has some nice symmetries with custom type support for bitfield members, just on the other end, making it possible to customize the representation of the `u32`.

Are there specific reasons for why you would prefer a wrapper type?

Bikeshedding: maybe `repr` would be a clearer name for the macro argument than `inner`, since this similar to Rust's `#[repr]` for enums.

closed 2024-06-02 by wrenger

mkroening, 2024-06-02:
> Well, I was just thinking about alternatives to keep this crate simple. The wrapper was just an Idea.

Sure, that makes sense. I hope I did not come across as too harsh. 😅

> If I understand you correctly, your goal is to do something alongside the following

Yes, that's correct. Ah, now I understand better what you meant. `Le<T>` actually is the generic type behind `le32`. `Le<MyBitfield>` would be a nice looking type, but I think it would be quite complex to implement.

Thanks a lot for accepting and for the wonderful crate! I will follow up with a documentation PR tomorrow. :)