wrenger/bitfield-struct-rs #43: Support spanning field that allows to be breaked in by other one.

magicwenli, 2024-07-18:
Assume we have a `u8` structure as below. And both bit `0..3` and `5..7` are defined by `Type`.

<table>
  <tr>
    <td>7</td>
    <td>6</td>
    <td>5</td>
    <td>4</td>
    <td>3</td>
    <td>2</td>
    <td>1</td>
    <td>0</td>
  </tr>
  <tr>
    <td colspan="3">Type</td>
    <td>P/R</td>
    <td colspan="4">Type</td>
  </tr>
</table>

I want to know if there's some good practice for this feature. Otherwise I'm willing to do some contribute.

For example something like this.

```rust
use bitfield_struct::bitfield;

#[bitfield(u8)]
struct MyByte {
    #[bits(3, take=type_)]
    type: usize,
    pr: bool,
    #[bits(4)]
    type_: usize
}

let my_byte = MyByte::new()
    .with_type(0b1111111)
    .with_pr(false);

assert!(my_byte.into_bytes(), 0b11101111);
```

wrenger, 2024-07-22:
Currently, there is no way to combine different bit ranges into a single member. Still, this is an edge case that comes up from time to time (e.g. Intel MMIO registers that have grown from 32 to 64 bit).

Currently, i just implement a custom function for this:

```rust
/// Describes the structure of segment descriptors
///
/// A data structure that contains size, position, access rights, and purpose of any segment.
/// Segment descriptors are used in both the GDT, as well as in LDTs.
///
/// - ISDMv3, 3.4.5; Segment Descriptors
/// - AAPMv2, 4.7 Legacy Segment Descriptors
#[bitfield(u64)]
#[derive(PartialEq, Eq)]
pub struct Descriptor {
    /// Least-significant bits of segment size (influenced by `pages`!)
    #[bits(16)]
    limit_low: u32,
    /// Least-significant bits of base address
    #[bits(24)]
    base_low: u32,
    /// Meaning of those 4 bits depends on `kind_code_data` below
    #[bits(4)]
    pub kind: u8,
    /// Descriptor type (influences the meaning of the 3 bits above)
    pub descriptor_kind: bool,
    /// Ring for this segment
    #[bits(2)]
    pub ring: Ring,
    /// Entry is valid iff set to `true`
    pub present: bool,
    /// Most-significant bits of segment size
    #[bits(4)]
    limit_high: u32,
    /// Bit which can be used for other purposes (in software)
    pub available: bool,
    /// Meaning of those 2 bits relate to `descriptor_kind` and `kind`
    #[bits(2)]
    pub custom: u8,
    /// Uses page granularity for the segment limit (instead of bytes)
    pub pages: bool,
    /// most-significant bits of base address
    base_high: u8,
}

impl Descriptor {
    /// Base/starting address
    pub const fn base(&self) -> u32 {
        self.base_low() | (self.base_high() as u32) << Self::BASE_LOW_BITS
    }
    /// Base/starting address
    pub const fn with_base(self, v: u32) -> Self {
        self.with_base_low(v & 0x00ff_ffff)
            .with_base_high((v >> Self::BASE_LOW_BITS) as _)
    }
    // ...
}
```

magicwenli, 2024-07-24:
Thank for your reply! I am agreeing with you this is an edge use case.

And it's not worthy to take lots of effort to implement it.

At the end, I managed to make it work by manually implementing all setter, getter and Debug traits.

Thanks again!

wrenger, 2024-07-26:
Ok, I hope it's not too much of a hassle. 

Still, I'm generally open for contributions. So if you find any other parts that could be improved, feel free to mention it.

closed 2024-07-26 by wrenger