wrenger/bitfield-struct-rs #44: No minimum supported Rust version is published.

EngJay, 2024-07-28:
I'm currently writing an [embedded-hal](https://github.com/rust-embedded/embedded-hal) driver that depends on this crate and had intended to match the MSRV of embedded-hal but did not find a published MSRV for this crate. It does not build with 1.60, the MSRV of embedded-hal, but does build with 1.65.

wrenger, 2024-08-04:
Yes, currently due to let-else we need 1.65.0, and I probably don't want to get stuck on too old compiler versions as it's harder to maintain :/

EngJay, 2024-08-04:
I wasn't suggesting supporting older versions, just that the MSRV be published so it doesn't have to be determined by dependents of your crate. Thanks for adding it!

EngJay, 2024-08-05:
I published the [initial release](https://crates.io/crates/pa-spl) of my driver that depends on this today - you're welcome to include it in any list of project the depend on yours. So far, this lib makes the most sense to me for how to represent registers of anything I've seen in Rust.

closed 2024-08-06 by wrenger