wrenger/bitfield-struct-rs #46: Question: validation before structure creation?

JPHutchins, 2024-08-11:
I have a simple protocol that defines 2 bits for its Version field.  Represented as an enum, only the values of 0 and 1 are valid - values 2 and 3 would represent a malformed message.  The following implementation works just fine for compile time safety, but I'm not sure how to handle the runtime checking of data on the wire that could be malformed.  Presently it hits `unreachable`.

What is ideal is for `unreachable` to be hit at compile time and for an Error to be generated at runtime.

Being new to Rust I am just stumped on how to do this declaratively!

The following is runnable with

```toml
[dependencies]
binrw = "0.14.0"
bitfield-struct = "0.8.0"
```

```rust
use binrw::{io::Cursor, BinRead, BinReaderExt, BinWrite};
use bitfield_struct::bitfield;

#[derive(Debug, PartialEq, Eq)]
#[repr(u8)]
enum Version {
    V1 = 0,
    V2 = 1,
}
impl Version {
    const fn into_bits(self) -> u8 {
        self as u8
    }
    const fn from_bits(bits: u8) -> Self {
        match bits {
            0 => Version::V1,
            1 => Version::V2,
            _ => unreachable!(),
        }
    }
}

#[bitfield(u8)]
#[derive(BinRead, BinWrite)]
struct Header {
    #[bits(3)]
    op: u8,
    #[bits(2)]
    version: Version,
    #[bits(3)]
    _reserved: u8,
}

fn main() {
    const GOOD_DATA: [u8; 1] = [0x01 | (1 << 3)];
    let mut reader = Cursor::new(&GOOD_DATA);
    let header: Header = match reader.read_be() {
        Ok(header) => header,
        Err(e) => panic!("Error reading header: {:?}", e),
    };
    println!("{:?}", header);

    const BAD_DATA: [u8; 1] = [0x01 | (2 << 3)];
    reader = Cursor::new(&BAD_DATA);
    let header: Header = match reader.read_be() {
        Ok(header) => header,
        Err(e) => panic!("Error reading header: {:?}", e),
    };
    println!("{:?}", header);
}
```

Output:
```
Header { op: 1, version: V2 }
thread 'main' panicked at src/main.rs:18:18:
internal error: entered unreachable code
```

So far I've added a "try from impl" to the Version that could return an `Option<Version>`.  But this is not right either, because the field of the struct is not Optional, it is an invariant of the struct.

Thanks for this great work!
JP

JPHutchins, 2024-08-11:
Also, it seems like Enums would be a common use case.  ChatGPT helped me generalize the into_bits and from_bits impls so that you can "derive bitfield" on enums.  This is where I was playing with using a try_from_bits to handle runtime checking by using Err or Option.

Anyway, I wonder if a feature like this would make since in this library?

```rust
extern crate proc_macro;
use proc_macro::TokenStream;
use quote::quote;
use syn::{parse_macro_input, Data, DeriveInput};

#[proc_macro_derive(Bitfield)]
pub fn bitfield_derive(input: TokenStream) -> TokenStream {
    let input = parse_macro_input!(input as DeriveInput);
    let name = input.ident;

    let gen = match input.data {
        Data::Enum(data_enum) => {
            let variants = data_enum.variants.iter().map(|v| {
                let ident = &v.ident;
                let discriminant = match &v.discriminant {
                    Some((_, expr)) => quote! { #expr },
                    None => quote! { unreachable!() },
                };
                quote! {
                    #discriminant => Self::#ident,
                }
            });

            let variants_checked = data_enum.variants.iter().map(|v| {
                let ident = &v.ident;
                let discriminant = match &v.discriminant {
                    Some((_, expr)) => quote! { #expr },
                    None => quote! { unreachable!() },
                };
                quote! {
                    #discriminant => Some(Self::#ident),
                }
            });

            quote! {
                impl #name {
                    const fn into_bits(self) -> u8 {
                        self as u8
                    }

                    const fn from_bits(bits: u8) -> Self {
                        match bits {
                            #(#variants)*
                            _ => unreachable!(),
                        }
                    }

                    const fn try_from_bits(bits: u8) -> Result<Self, Err> {
                        match bits {
                            #(#variants_checked)*
                            _ => Err(),
                        }
                    }
                }
            }
        }
        _ => panic!("Bitfield can only be derived for enums"),
    };

    gen.into()
}
```

JPHutchins, 2024-08-12:
I think that I have a solution after realizing that the OFFSET and BITS consts were generated by bitfield_struct!  Really cool!

```rust 
use bitfield_struct::bitfield;

#[derive(Debug, PartialEq, Eq)]
#[repr(u8)]
enum Version {
    V1 = 0,
    V2 = 1,
}
impl Version {
    const fn into_bits(self) -> u8 {
        self as u8
    }
    const fn from_bits(bits: u8) -> Self {
        match bits {
            0 => Version::V1,
            1 => Version::V2,
            _ => unreachable!(),
        }
    }
    fn try_from_bits(bits: u8) -> Result<Self, String> {
        match bits {
            0 => Ok(Version::V1),
            1 => Ok(Version::V2),
            _ => Err(format!("Invalid version: {}", bits)),
        }
    }
}

#[bitfield(u8)]
struct Header {
    #[bits(3, access = RO)]
    op: u8,
    #[bits(2, access = RO)]
    version: Version,
    #[bits(3)]
    _reserved: u8,
}
impl Header {
    fn try_from(data: u8) -> Result<Self, String> {
        Version::try_from_bits(
            (data >> Header::VERSION_OFFSET) & ((1u8 << Header::VERSION_BITS) - 1u8),
        )?;

        Ok(Header::from(data))
    }
}

fn main() {
    const GOOD_DATA: [u8; 1] = [0x01 | (1 << 3)];
    let header = Header::try_from(GOOD_DATA[0]);
    println!("{:?}", header);

    const BAD_DATA: [u8; 1] = [0x01 | (2 << 3)];
    let header = Header::try_from(BAD_DATA[0]);
    println!("{:?}", header);
}
```

I'll investigate getting this derived and I'd still love feedback since I don't know what I'm doing 🤣.

JP

wrenger, 2024-08-12:
Yes, I agree that input validation would probably be a good thing. Building your own `try_from_bits` functions is possible, but validation without panic currently is not. Ideally, one would design this around the `try_*` functions, meaning that `from_bits`/`into_bits` would call `try_from_bits`/`try_into_bits` and unwrap.

For the implementation, we would have two options: Either require all enums to implement the `try_from_bits`/`try_into_bits` functions (this would be cleaner but not backwards compatible) or having another argument in the `bits` attribute as shown below:

```rust
#[bitfield(u8)]
struct Header {
    #[bits(4)]
    normal_enum: NormalEnum,
    #[bits(4, try_from = TryEnum::try_from_bits, try_into = TryEnum::try_into_bits)]
    try_enum: TryEnum,
}

#[repr(u8)]
enum NormalEnum {
    A,
    B,
    C,
}
impl NormalEnum {
    fn from_bits(b: u8) -> Self {
        match b {
            1 => Self::B,
            2 => Self::C,
            _ => Self::A,
        }
    }
    fn into_bits(self) -> u8 {
        self as _
    }
}

#[repr(u8)]
enum TryEnum {
    A,
    B,
    C,
    D,
}
impl TryEnum {
    fn try_from_bits(b: u8) -> Result<Self, ()> {
        match b {
            0 => Ok(Self::A),
            1 => Ok(Self::B),
            2 => Ok(Self::C),
            _ => Err(()),
        }
    }
    fn try_into_bits(self) -> Result<u8, ()> {
        Ok(self as _)
    }
}
```

Still, I do not really like that we would have to ways to implement the bit conversion in this case. So I'm currently more in favor of breaking backwards compatibility and making the `from_bits`/`into_bits` functions return results (or options)...

wrenger, 2024-08-12:
Another nice use case would be to allow specifying your own validator functions like:

```rust
#[bitfield(u8)]
struct Header {
    #[bits(4, check = check_version)]
    version: u8,
    #[bits(4)]
    __: (),
}

fn check_version(version: u8) -> bool {
    version <= 3
}
```

wrenger, 2024-08-12:
Ok, I have to make a step back, this is not a simple problem, and I'm unsure if it really fits into this crate.

The general focus of this crate is to describe the memory layout of bitfields (for HW registers etc), not to parse and validate them. The conversion happens only when a generated memory function is called (`version`, `get_version`, `with_version`). Meaning that it would make more sense to generate `try_*` version of these functions.
The `struct`, e.g. `Header` from the previous example, is never constructed in this way, it just describes the memory layout of the underlying `u8`.

JPHutchins, 2024-08-12:
> The general focus of this crate is to describe the memory layout of bitfields (for HW registers etc), not to parse and validate them. The conversion happens only when a generated memory function is called (version, get_version, with_version).

Yeah, I like the idea of drawing the line there!

In fact, I continued getting so crazy with the typing that I had to move on to "wrapping" the bitfield_struct generated struct anyway.

JPHutchins, 2024-08-12:
BTW, I don't know how Discussions on GitHub work but if you're open to moving this there it might be more appropriate since there is no bug or feature request here.

wrenger, 2024-08-13:
> I'm hoping that this crate will work on a small embedded system. Features that can't be opted out of make that use case less practical. For example, the try_from_bits Err I made is not a const fn and uses a heap allocated string, which I might be annoyed with if I was on a small platform.

Yes, embedded, and OS dev were the original reasons for this crate, where you often just do not work with strings etc.

closed 2024-09-01 by wrenger