wrenger/bitfield-struct-rs #47: Field widths must be literal values - const values and expressions do not compile

DavidAntliff, 2024-08-16:
Please consider this struct:

```rust
#[bitfield(u32)]
struct Instruction {
    #[bits(20)]
    opcode: u32,
    #[bits(12)]
    __: u16,
}
```

The bitfield width is specified with a literal value of `20`, and the remaining "spare" bits have a literally specified width 12, to ensure the sum of the field widths equals the total bitfield width.

As things go, I want the opcode width to be determined by `const` code elsewhere, but to keep this issue simple, this does not work:

```rust
const INSTRUCTION_WIDTH: u32 = 32;
const OPCODE_WIDTH: u32 = 20;
const SPARE_WIDTH: u32 = INSTRUCTION_WIDTH - OPCODE_WIDTH;

#[bitfield(u32)]
struct Instruction {
    #[bits(OPCODE_WIDTH)]
    opcode: u32,
    #[bits(SPARE_WIDTH)]
    __: u16,
}
```

The compiler error is:

```
error: expected `=`
   --> src/emitter/encoding.rs:140:1
    |
140 | #[bitfield(u32)]
    | ^^^^^^^^^^^^^^^^
    |
    = note: this error originates in the attribute macro `bitfield` (in Nightly builds, run with -Z macro-backtrace for more info)

error: malformed #[bits] attribute
   --> src/encoding.rs:142:5
    |
142 |     #[bits(OPCODE_WIDTH)]
    |     ^
```

It looks like only literals are supported.

This is essentially requiring the use of "magic numbers", which is not a good practice in software engineering, and leads to duplication of values (violating the "DRY" principle). Collectively, this leads to more fragile software.

----

Taking another approach, maybe we can have the bitfield-struct as the authoritative source of the bitfield widths?

Is there a way to statically extract the size of each field, so that something like this is possible?

```rust
const OPCODE_WIDTH: u32 = Instruction::opcode::width;

fn encode(v: f32) {
    let x = v * (1 << OPCODE_WIDTH) - 1;

    let instruction = Instruction::new().with_opcode(x);

    // ...
}
```

----

There's another case that doesn't work either, where the backing type is a type alias. Consider this example where the bitfield-struct is intended to support any opcode width from 1 to 32, if the above did work:

```rust
type INS_TYPE = u32;
const INS_WIDTH: u32 = 8 * std::mem::size_of::<INS_TYPE>() as u32;
const OPCODE_WIDTH: u32 = 20;
const SPARE_WIDTH: u32 = INS_WIDTH - OPCODE_WIDTH;

#[bitfield(INS_TYPE)]
struct Instruction {
    #[bits(OPCODE_WIDTH)]
    opcode: INS_TYPE,
    #[bits(SPARE_WIDTH)]
    __: INS_TYPE,
}
```

This fails because the macro is checking the literal rather than the type it resolves to:
```
error: unsupported type
   --> src/encoding.rs:143:1
    |
143 | #[bitfield(INS_TYPE)]
    | ^^^^^^^^^^^^^^^^^^^^^
```

If this kind of code was supported, it would significantly help with building robust and self-vaidating bitfield representations.

wrenger, 2024-08-17:
Currently, the macro uses the bit sizes directly for bounds checking. External constants (especially ones that depend on type sizes) cannot be evaluated at the time of macro expansion and thus are not supported for now.

Generally, this might be possible to implement (by generating const asserts, etc.), but is definitely not that easy.

DavidAntliff, 2024-08-17:
Fair enough, although the lack of a single source of truth still bothers me.

Thus, is it possible for these values (field widths) to be extractable from the bitfield-struct itself, as compile-time consts? Could the macro construct public associated consts for each field (if it doesn’t already)?

Incidentally, can they be extracted at runtime? That might be sufficient in many cases.

wrenger, 2024-08-18:
Yes, the macro constructs two constants for each field (*_OFFSET for the bit offset in the bitfield and *_BITS for the bit size).

DavidAntliff, 2024-08-18:
Ok, thank you, that's useful to help reduce some fragility:

Defined elsewhere:
```rust
// Used by other code that does *not* depend on the bitfield-struct, it only needs to know the maximum representable value:
const OPCODE_WIDTH: usize = 20;
```

Defined in the encoding implementation:
```rust
#[bitfield(u32)]
struct Instruction {
    #[bits(20)]
    opcode: u32,
    #[bits(12)]
    __: u16,
}

// Check our assumptions are correct:
const_assert_eq!(
    Instruction::OPCODE_BITS,
    OPCODE_WIDTH,
);
```

Although the direction of the flow of information is still not quite right as two or more bitfield-structs could reuse a field width. Otherwise I could refactor to write:

```rust
const OPCODE_WIDTH: usize = Instruction::OPCODE_BITS;
```

But if there are multiple structs defining the same bitfield then I'd have to pick one and hope/assert all of the others used the same value. In the end, I think a battery of static assertions for each bitfield-struct is the safest thing.

I notice that the `#[bits(..., default=X)]` can take a `const` as a parameter. I'm still not 100% clear on why this is OK for the default, but not for the bitfield width.

wrenger, 2024-08-23:
Yes, the default expressions are only copied and pasted into the new constructor. The bit sizes are, however, used for calculations and overflow checks...

wrenger, 2024-09-01:
> Yes, the default expressions are only copied and pasted into the new constructor. The bit sizes are, however, parsed and used for calculations and overflow checks...

Maybe I should explain this more. Macros in rust are expanded after tokenization, this means the macro only knows the input tokens. It has no knowledge of what is behind variables, so we cannot use any constants as they are not evaluated yet by the compiler. The const evaluation comes afterward. The macro even has no or only limited information about the target platform the code is compiled for. So even "simple" things like the bit size of a `usize` is not known or if the platform supports `u128` and so on.

Still, there are some workarounds, like generating code that contains `#[cfg(...)]` attributes, just not depending directly on input expressions and just pasting them as they are into the generated code, letting the rust compiler evaluate them later...
Still, it is quite difficult to get meaningful error messages this way, e.g. if the bit size of a field is larger than its type or the entire bitfield...
And we could not include the bit offsets in the documentation, etc.

DavidAntliff, 2024-09-01:
That makes sense - thank you for explaining.

closed 2024-09-01 by DavidAntliff