wrenger/bitfield-struct-rs #48: Incorrect doc descriptions for bitfield widths (off by 1)

DavidAntliff, 2024-08-18:
Please consider:

```
#[bitfield(u32)]
struct Instruction {
    #[bits(20)]
    opcode: u32,
    #[bits(1)]
    ro: u8,
    #[bits(11)]
    __: u16,
}
```

In the macro expansion, this results in:

```
...

    const OPCODE_BITS: usize = 20usize;
    const OPCODE_OFFSET: usize = 0usize;
    #[doc = "\n\nBits: 0..20"]
    const fn opcode(&self) -> u32 {

...

    #[doc = "\n\nBits: 20..21"]
    #[cfg_attr(debug_assertions, track_caller)]
    const fn with_ro(self, value: u8) -> Self {

...

    #[doc = "\n\nBits: 20..21"]
    #[cfg_attr(debug_assertions, track_caller)]
    fn set_ro(&mut self, value: u8) { *self = self.with_ro(value); }

...
```

"0..20" is 21 bits, and "20..21" is 2 bits, but the associated field widths are 20 and 1, respectively. 

Thus I think these `#[doc ...]` entries should read "Bits: 0..19" and "Bits: 20" (or "Bits: 20..20" if it must) respectively.

bitfield-struct = "0.6.1"

wrenger, 2024-08-20:
In Rust, `0..20` is an exclusive range, meaning that it goes from 0 to 19.
Alternatively, we could have chosen an inclusive `0..=20`, which contains the 20, but this is more a style choice.

DavidAntliff, 2024-08-20:
Ah, good point, I hadn’t considered the docs were in Rust notation with two dots. Too much time spent looking at manufacturers’ data-sheets perhaps! (None of which have ever used this right-exclusive interval notation, I might add).

In the context of Rust I agree that it makes sense as-is.

(A far more common notation in data-sheets would be 0-19 or 0…19 with three dots).

DavidAntliff, 2024-08-20:
FWIW, I think “0..=19” would actually be clearer for anyone reading the generated docs without Rust knowledge, which is more common than you might think. If the bitfield-structs are authoritative then the docs generated from them may go as a “data-sheet” into non-Rust contexts where the double-dot could be misinterpreted, as I have done here.

closed 2024-08-20 by DavidAntliff

wrenger, 2024-08-21:
Yes, that is a good point. `0..=19` might be clearer.