wrenger/bitfield-struct-rs #49: Feature request: checked field-setting functions for opt-in bounds checking in release builds

DavidAntliff, 2024-09-09:
Please refer to conversation #47 for background.

For a debug/dev build, there is a `debug_assert!()` call on the value of `with_<fieldname>()` to ensure that the supplied value is within bounds. If the value is outside of allowed bounds, the thread panics. This is good.

For a release build, this check is disabled, which is also good, usually.

However, I can't see an easy way for code calling this function to do the bounds check itself, easily, without replicating the logic within the generated `with_<fieldname>()` method.

Could this be included in an additional generated function, say, `with_<fieldname>_checked()`, that returns a `Result`?

For now, I am doing this manually on every call to `with_<fieldname>()` using `*_BITS`, but it's easy to accidentally specify the wrong constant:

```rust
    fn check(value: u8, field_bits: u8) -> Result<(), Error> {
        let this = value;
        let value: u8 = this as _;
        let mask = u8::MAX >> (u8::BITS - field_bits as u32);
        if value <= mask { Ok(()) } else { Err(Error(...)) }
    }

    Instruction::new().with_opcode(check(value, Instruction::OPCODE_BITS)?);
    Instruction::new().with_length(check(value, Instruction::LENGTH_BITS)?);
    Instruction::new().with_command(check(value, Instruction::LENGTH_BITS)?);  // Oops! Wrong field length!
```

There's a bit too much DRY violation here.

This would be nicer and less error-prone:

```
    Instruction::new().with_opcode_checked(value)?;
    Instruction::new().with_length_checked(value)?;
    Instruction::new().with_command_checked(value)?;
```

wrenger, 2024-09-12:
I agree, this is generally a good idea. I hope I find some time in the next days to implement this.

wrenger, 2024-09-23:
So I finally got the time to implement this (b636181e7e157ecd667d5701ceae14c4c1e6adf6).

I'm a bit unsure what to put into the Error. Ideally we could create a custom OverflowError type. But this crate is currently a single proc_macro, which cannot provide any additional types/functions, only macros. I would have to create another crate for that and restructure the repo...

Alternatively we could stick to `()` or a core library type...

DavidAntliff, 2024-09-23:
Great news - thank you.

> Alternatively we could stick to () or a core library type...

I think `()` is probably OK, and if documented then it's fairly clear what it signals. There's really only one way the function can fail, and no context is needed to interpret that. A user can always call `.map_err(|_| ...)` on the result, if they need to add more error information.

wrenger, 2024-09-23:
Ok I've published this as 0.9.0

DavidAntliff, 2024-09-24:
Great, seems to be working - thank you! :)

closed 2024-09-24 by DavidAntliff