wrenger/bitfield-struct-rs #53: fields with prefix `_` give `warning: manual implementation of an assign operation` in clippy since 0.9.1

s5suzuki, 2024-10-14:
```
#[bitfield(u8)]
struct MyByte {
    #[bits(4)]
    kind: usize,
    #[bits(4)]
    _pad: usize,
}
```
This code will cause a `warning: manual implementation of an assign operation` in clippy.
Changing `_pad` to `pad`, it works fine.

wrenger, 2024-10-14:
Oh ok, probably a regression from 3f48bdc573f3b230330253d3215f15fb8a567ffc

Can you tell me the name of the clippy check.

s5suzuki, 2024-10-14:
Yes, of course, the name is `clippy::assign_op_pattern`.

The following is a full log of the error.
```
error: manual implementation of an assign operation
 --> src/main.rs:3:1
  |
3 | #[bitfield(u8)]
  | ^^^^^^^^^^^^^^^
  |
  = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#assign_op_pattern
  = note: `-D clippy::assign-op-pattern` implied by `-D warnings`
  = help: to override `-D warnings` add `#[allow(clippy::assign_op_pattern)]`
  = note: this error originates in the attribute macro `bitfield` (in Nightly builds, run with -Z macro-backtrace for more info)

error: manual implementation of an assign operation
 --> src/main.rs:3:1
  |
3 | #[bitfield(u8)]
  | ^^^^^^^^^^^^^^^
  |
  = help: for further information visit https://rust-lang.github.io/rust-clippy/master/index.html#assign_op_pattern
  = note: this error originates in the attribute macro `bitfield` (in Nightly builds, run with -Z macro-backtrace for more info)
```

s5suzuki, 2024-10-14:
Sorry, I accidentally closed the issue.

wrenger, 2024-10-15:
Fixed in 38054deedb43455e4d7d8f55f28113f091f817ee and released as 0.9.2

closed 2024-10-15 by wrenger