wrenger/bitfield-struct-rs #54: Failable conversion?

ianthetechie, 2024-10-19:
Hey, thanks for the great crate!

It would be neat if there was a failable version, for cases when an enumeration with an integral `repr` doesn't use the full range (some bit patterns are invalid). I'm not sure if this fits within the goals of the crate, but thought I'd open the feature request anyways.

wrenger, 2024-10-29:
Apologies for the delayed response. So you are looking for getters that return a result?

```rust
enum MyEnum {
    A = 1,
    B = 2,
    // others are not valid
}

#[bitfield(u8)]
struct Bitfield {
    #[bits(4)]
    my_data: MyEnum,
    #[bits(4)]
    other: u8,
}

assert_eq!(Ok(MyEnum::B), Bitfield(0xffff_0002).my_data_checked());
assert!(Bitfield(0xffff_ffff).my_data_checked().is_err());
```

mehmetalianil, 2025-05-24:
Not the owner, but I also wished this. 
In many packet structures there are ranges of reserved values that might be omitted. 
Now when I write something like this: 
```rust
#[bitfield(u64)]
pub struct BitfieldExample {
    first: u8,
    #[bits(4)]
    second: u8,
// . . .
    #[bits(5)]
    third_enum: MyEnum,
    #[bits(3)]
    __: u8,
}

#[derive(Default, PartialEq)]
#[repr(u8)]
pub enum MyEnum {
    FirstState = 0,
    SecondState = 2,
    ThirdState = 7,
    FourthState = 8,
    //... 
    #[default]
    LastState = 31,
}
```

Rust analyzer asks for implementations for `from_bits` and `into_bits` as:
```
no variant or associated item named `from_bits` found for enum `MyEnum` in the current scope
```

I cannot really implement these functions because they are infallible, their fn signatures do not return Option or Result.