wrenger/bitfield-struct-rs #55: From bits should respect padding

gregorygaines, 2024-11-28:
If I use `from_bits(0xFF)`, and I have padding fields, I expect from_bits to ignore padding bit fields to properly represent the struct

gregorygaines, 2024-11-29:
For example:

```
 #[bitfield(u8)]
    struct MyBitfield {
        #[bits(default = true)]
        flag: bool,
        #[bits(2, default = 0b11)]
        data: u8,
        /// Padding
        #[bits(5)]
        __: u8,
    }

    let mut val = MyBitfield::from_bits(0xFF);

    let raw: u8 = val.into();
    println!("{raw:b}");
    ```
    
    This returns `11111111` when it should be `00000111`.

wrenger, 2024-11-29:
Padding bit are not touched by the macro. In this example, they remain set to 1.

It looks like you want to overwrite the padding bits to 0?

gregorygaines, 2024-11-29:
Yes, with padding bits, I expect them to stay zero even when I use from_bits.

gregorygaines, 2024-11-30:
I tried my hand at a PR, #58.

wrenger, 2024-12-03:
I have to admit that I do not really support this change. There are a lot of hardware registers where you are not allowed to change padding/reserved values. So this change would break the read-modify-update cycle on the registers:

```rust
/// LApic timer control register, R/W
#[bitfield(u32)]
struct TimerCtrl {
    /// This vector in the IDT will be activated for the timer interrupt.
    vector: u8,
    #[bits(4)]
    __: (),
    /// Whether the interrupt delivery is pending (true) or idle.
    delivery_status: bool,
    #[bits(3)]
    __: (),
    /// Mask (disables) or unmask timer interrupts.
    masked: bool,
    /// Timer mode.
    #[bits(2, default = TimerMode::OneShot)]
    timer_mode: TimerMode,
    #[bits(13)]
    __: (),
}

let mut reg = TimerCtrl::from_bits(ptr.read_volatile());
reg.set_masked(false); // everything else (especially padding) remains the same
ptr.write_volatile(reg.into_bits());
```

closed 2024-12-03 by gregorygaines