wrenger/bitfield-struct-rs #60: Warnings for `clippy::unnecessary_cast` for default values

martiege, 2025-01-13:
Thank you for this great package! I've been using it with success, but it generates some warnings for `clippy::unnecessary_cast`, e.g.

```
 28 |   #[bitfield(u16)]
    |  ____________^
 29 | | #[derive(PartialEq, Eq)]
 30 | | pub struct HardwareVersion {
 31 | |     #[bits(4, access = RO)]
 32 | |     pub hardware_version: u8,
 33 | |
 34 | |     #[bits(12, default = 0_u16)]
    | |______________________________^ help: try: `0_u16`
```

wrenger, 2025-01-13:
Ah interesting. The cast is usually only necessary if the bitfield type differs from the struct field type.

martiege, 2025-01-14:
Yes, the strange thing is that explicitly casting does remove the warning for some data structures, but not from others... 

This gives the warning:
```rust
#[bitfield(u16)]
#[derive(PartialEq, Eq)]
pub struct HardwareVersion {
    #[bits(4, access=RO)]
    pub hardware_version: u8,

    #[bits(12, default=0_u16)]
    __: u16,
}
```

While this does not:
```rust
#[bitfield(u16)]
#[derive(PartialEq, Eq)]
pub struct CalibrationDataChars {
    #[bits(10, access=RO)]
    pub calibration_data_chars: u16,

    #[bits(6, default=0_u8)]
    __: u8,
}
```

And just for fun, this gives the warning (but specifically for u16, not u8 as the empty bits should be):
```rust
#[bitfield(u16)]
#[derive(PartialEq, Eq)]
pub struct CalibrationDataChars {
    #[bits(10, access=RO)]
    pub calibration_data_chars: u16,

    #[bits(6, default=0)]
    __: u8,
}
```

I'm guessing this happens due to some internal way this structure is represented. The fact that the first data structure is a u16 bitfield, and the empty bits are also u16, while the second structure is a u16, and the empty bits are u8, is also likely related somehow.

wrenger, 2025-01-15:
Ok I'll look into this :)

wrenger, 2025-01-26:
Fixed in 0.9.4 and 0.10.1

closed 2025-01-26 by wrenger