wrenger/bitfield-struct-rs #61: Rust MSRV jump

kevinhartman, 2025-01-16:
Hi there, and thanks a lot of this neat library!

I don't know that this makes sense as an "issue", but the latest minor version (0.10.0) jumps the Rust MSRV from 1.65 to 1.83. As I'm sure you're aware, any crates depending on 0.10.0 will now need to update their MSRV to 1.83. I was hoping to use this crate in Qiskit for our instruction representation, but we cannot raise our MSRV to 1.83 yet (we are currently at 1.70, and discussing a bump to 1.79). That said, we don't strictly _need_ 0.10.0, but we tend to be somewhat conservative, so I hesitate to add a dependency on `bitfield-struct` if we will need to aggressively update our own MSRV to pick up bugfixes etc.

These stats recently came to my attention, which seem helpful when planning bumps.
https://lib.rs/stats#rustc 

Feel free to close this whenever!

wrenger, 2025-01-17:
The jump was motivated by the new support for const mut references in 1.83. I have been waiting for this for a long time. 
Still, I understand the need for backwards compatibility. I have to find out if this could be enabled conditionally based on the compiler version. If that is the case, I could lower the version requirement again...

wrenger, 2025-01-26:
I'll maintain the `0.9` branch for the ongoing future, and have backported your PR #59.

Sorry that it took so long. Last week was just too busy.

kevinhartman, 2025-02-05:
Thanks a lot for doing that, it's much appreciated.

I was indeed able to use your library in Qiskit (though I had to concede on using some of its cooler features).

Looking forward to seeing how this work evolves 🙂.

closed 2025-02-05 by kevinhartman