wrenger/bitfield-struct-rs #62: Hashing of bitfield structs with unused space/padding

freemin7, 2025-03-05:
Assuming i have stuct with padding.
```
#[bitfield(u64)]
#[derive(PartialEq, Eq)]
struct MyBitfield {
    int: u16,
    #[bits(default = true)]
    flag: bool,
    #[bits(1)]
    tiny: u8,
    #[bits(13)]
    negative: i16,
    #[bits(16)]
    custom: CustomEnum,
    #[bits(10)]
    pub public: usize,
    #[bits(1, access = RO)]
    read_only: bool,
    #[bits(1, access = WO)]
    write_only: bool,
    /// Padding
    #[bits(5)]
    __: u8,
}
```

It is currently not documented whether hashing done on this type will depend on the data in padding or not. I think hashing is something that should be considered and documented. As i am new to Rust i am not confident that i can come to any conclusion on this. If my experimentation yields a definite result i will share my findings.

wrenger, 2025-03-06:
Until now, I've not really thought about hashing. Yes, the hasher uses the internal representation that includes padding.
Padding on bitfields is zero initialized when calling `new()`, if not specified otherwise (`#[bits(N, default=...)]`). 
If the bitfield is created with `from_bits()`, the padding is **not** touched or modified in any way. All setters only modify their respective bits.

The desired behavior would be to ignore padding or not?

freemin7, 2025-03-09:
For my purposes the desired behaviour would be to ignore the data in the padding.
Since your libraries makes no guarantees on what happensto data in the padding i don't think that behaviour would be suprising.
My prefered solution for ignoring the padding would be to and the data with a mask which zeros the padding before hashing but a feed each field into the hash function approach could also work (is slower though, as more hashing is done).

Usecases where the padding is filled by extra data, which should show up in the padding seem scarce. As the developer of the bitfield struct you always have a better solution then abusing the padding. Abusing the padding might be interesting for third parties which want to hijack libraries in ways developer doesn't intend but those are the only group of people i can think of. Padding data being ignored would bite them if the abused padding structs are feed into hash tables/bloom filter whose implementation they can not change either.

I'd suggest zeroing the padding before hashing but i won't push for considering zeroing in from_bits, that's a library wide decision.

wrenger, 2025-05-04:
I've now implemented an extra parameter for generating hash implementations that ignore padding [44acf33](https://github.com/wrenger/bitfield-struct-rs/commit/44acf339a1ef00384a61d74f2a11b05dd10a9318)

This can be enabled with:

```rust
#[bitfield(u32, hash = true)]
struct MyBitfield {
    data: u16,
    __: u8,
    #[bits(8)]
    extra: u8,
}
```

closed 2025-05-04 by wrenger